ArminJo/Smart-Battery-Module-Info_For_Arduino #4: [Feature Request] Permanent Failure Clear key

JyeSmith, 2019-05-20:
Hi, this is a great tool and instantly showed what is wrong with a couple of my batteries.

> Manufacturer Status (BIN): 0b10111001
> - FET Status 2
> - State: 0b1001
>  - Permanent Failure
>  - PFStatus: 0b100

Ive found in http://www.ti.com/lit/er/sluu313a/sluu313a.pdf on page 77 **B.16 PFKey (0x62)** that the permanent failure can be cleared and hopefully the batteries will be ok to use... finger crossed!

Are you able to assist with a script to clear the PF flag?  or help with a few pointers?

Cheers

ArminJo, 2019-05-20:
Hi Jye
it would be a challenge to unlock a device, but I would definitely give it a try. 
Do you have an idea what the password(s) could be.
Maybe I could integrate a automatic unlock and PF reset function using a few standard passwords.
And maybe we could use email for the development :-)

JyeSmith, 2019-05-20:
Ah, I see a password is required for full access.

I don't have the passwords but a script could be written to brute force it.  It could take a while but just loop through all combination until the full access flag is set.

Do you think that's possible with your current library?  I have a half a dozen different batteries that could be tested and used to start gathering a library.  This could be super helpful!!!

JyeSmith, 2019-05-20:
I see the access key is 2 words (32 bit)!  I think we are stuffed :(

ArminJo, 2019-05-20:
Did you check the fuses on the batteries, they might be fused if permanent failure is set and then you have an hardware problem!
But maybe you could ask google for default passwords. 
Otherwise you need 4 seconds between each password try-:(

closed 2019-10-15 by ArminJo