ArminJo/Smart-Battery-Module-Info_For_Arduino #9: Compilation error

N0ury, 2023-08-25:
/Users/nb/Sync/platformio/projects/tests/battery_checker/Smart-Battery-Module-Info_For_Arduino/SBMInfo/SBMInfo.ino: In function 'void setup()':
/Users/nb/Sync/platformio/projects/tests/battery_checker/Smart-Battery-Module-Info_For_Arduino/SBMInfo/SBMInfo.ino:357:26: error: 'getVCCVoltageMillivolt' was not declared in this scope
     myLCD.print(((float) getVCCVoltageMillivolt()) / 1000, 2);
                          ^~~~~~~~~~~~~~~~~~~~~~
/Users/nb/Sync/platformio/projects/tests/battery_checker/Smart-Battery-Module-Info_For_Arduino/SBMInfo/SBMInfo.ino:357:26: note: suggested alternative: 'sDesignVoltageMillivolt'
     myLCD.print(((float) getVCCVoltageMillivolt()) / 1000, 2);
                          ^~~~~~~~~~~~~~~~~~~~~~
                          sDesignVoltageMillivolt

exit status 1

Compilation error: 'getVCCVoltageMillivolt' was not declared in this scope

Board choosen: Arduino nano (to avoid any issue)
My goal is to use an ESP32-S2 or ESP32-C3

animevietsub, 2023-08-25:
Make sure that you have the file `MeasureVoltageAndResistance.hpp`, `ADCUtils.h`, `ADCUtils.hpp` in your project.

I see this project is not compatible with ESP32 yet, If you want it to work you need to change the `ADCUtils.h` and `ADCUtils.hpp` because it is designed to work with AVR microcontrollers.

N0ury, 2023-08-25:
> Make sure that you have the file `MeasureVoltageAndResistance.hpp`, `ADCUtils.h`, `ADCUtils.hpp` in your project.

They are there.
I've done as asked in the README.
git clone the_project
Then in Arduino IDE I've compiled the project just cloned (SBMInfo directory). Nothing less and nothing more.
I tried in platformIO, same issue.
Of course I've used Arduino nano as target because I had many other errors with ESP32.

N0ury, 2023-08-25:
Adding #include "ADCUtils.hpp" in SBMInfo.ino solved the problem.
I close the issue.

closed 2023-08-25 by N0ury